google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu7t5v0 #23: gf180mcu_fd_sc_mcu7t5v0_mux2_1 is seemingly not DRC clean

gatecat, 2022-11-02:
## Expected Behavior
All standard cells pass DRC

## Actual Behavior
`gf180mcu_fd_sc_mcu7t5v0_mux2_1` fails DRC with 2 of the following errors

```
Metal1 overlap of contact < 0.055um in one direction (CO.6)
```

[image]

## Steps to Reproduce the Problem

1. Build a design that instantiates `gf180mcu_fd_sc_mcu7t5v0_mux2_1` (I think it isn't used by synthesis automatically)
1. Observe DRC failures in Magic
1. Load the `gf180mcu_fd_sc_mcu7t5v0_mux2_1` to find it is the culprit

## Specifications

- Version: `open_pdks 120b0bd69c745825a0b8b76f364043a1cd08bb6a`
- Platform: Arch Linux

vijayank88, 2022-11-08:
@gatecat 
Can you add `gf180mcu_fd_sc_mcu7t5v0_mux2_1` into `no_synth.cells` and `drc_exclude.cells` and try?

mithro, 2022-11-28:
PTAL - @atorkmabrains / @mkkassem

mithro, 2022-11-28:
This is obviously related to https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu7t5v0/issues/2

atorkmabrains, 2022-11-29:
@RTimothyEdwards Could you check that?

RTimothyEdwards, 2022-11-29:
@atorkmabrains :  I don't have a rule in magic that exactly implements the GF CO.6 rule.  I would have to work out a much more complicated implementation of the rule.  The rule is not exactly "overlap >= 0.06um in one direction" but is actually "overlap >= 0.06um in one direction if overlap < 0.04 in the other direction".  There should be a way to implement this rule but I'll have to give it some thought.  The klayout DRC deck should indicate that it's clean.

atorkmabrains, 2022-11-29:
@gatecat I advise that you run this design klayout DRC and see if it's violating.

atorkmabrains, 2022-11-29:
@RTimothyEdwards Thanks for indicating this.

atorkmabrains, 2023-04-10:
@gatecat I'll close this issue as there is no further communications. Please re-open if you still need any thing.

closed 2023-04-10 by atorkmabrains